mzahana/px4_fast_planner #1: Required interface for integrating into mavros_controllers

Jaeyoung-Lim, 2020-06-26:
Hi, 

This looks awesome! I was looking at what you needed to change on the `mavros_controllers` and saw that you added a interface for `quadrotor_msgs/PositionCommand` message.

Can we maybe try to bring this in to `mavros_controllers`? If so how should we do this? Is there any chance where we can convert this into one of the standard messages?

mzahana, 2020-06-26:
Thanks @Jaeyoung-Lim !
I actually made changes in both Fast-Planner and mavros_controllers.

For mavros_controllers
* Added a [callback](https://github.com/mzahana/mavros_controllers-1/blob/fast_planner_interface/geometric_controller/src/geometric_controller.cpp#L75-L89) which subscribes to reference trajectory from Fast-Planner. Fast-Planner publishes reference trajectory to a topic with custom message [quadrotor_msgs/PositionCommand](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/master/uav_simulator/Utils/quadrotor_msgs/msg/PositionCommand.msg)

So, to standardize this, modifications would need to be made in both packages, or at least modify Fast-Planner to publish additional topic with standard message type.

In any case, what standard message type you think can be used here? We will need a message that has 3D vector for reference position, velocity, acceleration, yaw, and yaw_dot.

Jaeyoung-Lim, 2020-06-26:
Okay, so I think we have two options

- We add a node that republishes the message from `quadrotor_msgs/PositionCommand` to [`controller_msgs/flatTarget`](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/controller_msgs/msg/FlatTarget.msg) and it is included in this repo

- We add a interface for a standard message and add a node in this repo which translates `quadrotor_msgs/PositionCommand` to one of the standard messages and add the interface to `geometric_controller` 

what do you think?

mzahana, 2020-06-26:
Those are two good options, and I was already thinking about the second one. I prefer the second one (having interface to some standard ROS message) because it makes sub-packages more portable. For example, any planning package can just use a standard message without worrying about what controllers will be used. The question is which message would be suitable in this case?

Jaeyoung-Lim, 2020-06-26:
The standard interfaces that geometric_controller already supports are:
- multiDOFJointTrajectory message: https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L134
- TwistStamped https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L74 with a separate yaw target https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L130 with `Float32`

mzahana, 2020-06-26:
Although multiDOFJointTrajectory is more than enough since it is for an array for point while we need only one,  it seems like this what we can go with for now since it has rotational components from which we can extract target yaw directly and not relying on a separate topic for it. I will implement a python node for it and get back to you.

mzahana, 2020-06-26:
@Jaeyoung-Lim  [here is a PR](https://github.com/mzahana/px4_fast_planner/pull/3) for a node that converts from `PositionCommand` to `multiDOFJointTrajectory`. This also requires [this PR](https://github.com/Jaeyoung-Lim/mavros_controllers/pull/131) on mavros_controllers side, to work

Jaeyoung-Lim, 2020-07-03:
Closing since fixed in https://github.com/Jaeyoung-Lim/mavros_controllers/pull/131

closed 2020-07-03 by Jaeyoung-Lim