mzahana/px4_fast_planner #2: Change camera parameters

lgh5054, 2020-06-26:
Hello

It's so exciting!

I also want to see the fast-planner 3 days ago and incorporate it into real hardware.

The T265 slam the role of GPS and wants to implement avoidance with the D435

mzahana, 2020-06-26:
Hi @lgh5054 

You can change the depth topic using the `depth_topic` argument  in the `px4_fast_planner` [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L16)

You also need to specify the camera pose topic using the `camera_pose_topic` argument [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L15). This is the camera pose in the `world` frame

There is a node that looks for the transform from the camera frame to the world frame, [see here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L55-L61)
So, it's required to have transforms
* from `camera_link` to `base_link`. This is already done [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L23)
* and from `base_link` to `map`. This is done in the px4 config file [here](https://github.com/mzahana/px4_fast_planner/blob/master/config/px4_config.yaml#L69)
* and from `map` to `world`. This is done [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L22)

I hope this helps

mzahana, 2020-06-26:
Also pay attention to the camera intrinsic parameters, [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_topo_replan.launch#L33-L36)

lgh5054, 2020-06-27:
@mzahana Thank you for your answer.

I'll test it and share the results.

skyaid, 2020-06-30:
@mzahana 
[image]

As I said, I changed all parameters, but the screen does not print depth

mzahana, 2020-06-30:
@skyaid 
No depth will be printed on screen. Instead, you should see point cloud in Rviz. Can you share your tf tree `rosrun rqt_tf_tree rqt_tf_tree` ?

skyaid, 2020-06-30:
@mzahana 
[image]

mzahana, 2020-06-30:
@skyaid You have some issues with your setup.
* Your tree is not well connected
    * It seems that you are not running mavros using launch file that is included in `px4_fast_planner.launch`. It calls `mavros.launch` which uses a custom px4 plugin yaml file. not the default one that comes with mavros
    * The `camera_pose_publisher.py` node is looking for a transformation from `parent_frame` (which is set to `worls` in your file) to `child_frame` (which is set to `camera_link`. However, there is no connection between them! You need to make a connection to the word frame in your tree.
* I think `<arg name="camera_pose_topic" default="/camera_link"/>` is wrong. This should be set to the topic that will be published by the `camera_pose_publisher.py` node which in this file is `camera/pose`

skyaid, 2020-07-01:
@mzahana Thank you for your prompt reply

There has been a lot of progress

just like you said.

using px4_fast_planner px4.launch and change camera_pose_topic" default="/camera/pose

but tf not found in rviz 
/mavros/local_position/odom 
/mavros/local_position/pose   it's being published

[image]
"/>

[image]

mzahana, 2020-07-01:
@skyaid Please look at the `px4_fast_planner px4.launch` file. You will see that it already includes `geometric_controller.launch`

closed 2020-07-03 by lgh5054

atmaca11, 2021-05-06:
@lgh5054 Can you share your e-mail address ?