mzahana/px4_fast_planner #4: Resource not found: The following package was not found in <arg default="$(find mavlink_sitl_gazebo)/worlds/empty.world" name="world"/>: mavlink_sitl_gazebo

Muzaib95Anjum, 2020-07-06:
After following all the instructions when i run roslaunch px4_fast_planner px4_fast_planner.launch. i get the errors:
Resource not found: The following package was not found in <arg default="$(find mavlink_sitl_gazebo)/worlds/empty.world" name="world"/>: mavlink_sitl_gazebo
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/muzaib/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ROS path [3]=/home/muzaib/Firmware
The traceback for the exception was written to the log file

mzahana, 2020-07-06:
@Muzaib95Anjum looks like you did not includ PX4 SITL packages on the path. Did you do the setup using the `setup.sh` script in the [install](https://github.com/mzahana/px4_fast_planner/tree/master/install) folder? It already does that for you and the necessary lines in your `.bashrc` file

Muzaib95Anjum, 2020-07-06:
@mzahana yes i did setup using setup.sh script in the install folder! But the error remains....

mzahana, 2020-07-06:
@Muzaib95Anjum Are the paths updated in you `.bashrc` Also make sure that you source your `.bashrc` before you launch the setup?
```sh
cd
source .bashrc
```

Muzaib95Anjum, 2020-07-06:
@mzahana let me check again then tell you the situation.

mzahana, 2020-07-06:
@Muzaib95Anjum I just noticed that there are few steps missing in the `setup.sh` file. I will update it.

Muzaib95Anjum, 2020-07-06:
@mzahana okay when you update the the file kindly do notify.

mzahana, 2020-07-06:
@Muzaib95Anjum `setup.sh` file is updated. Kindly re-run it and check.

Muzaib95Anjum, 2020-07-06:
@mzahana okay checking it.

mzahana, 2020-07-06:
Also make sure that the end of your `.bashrc` looks like the following
```sh
source /opt/ros/melodic/setup.bash
source /home/arrow/catkin_ws/devel/setup.bash
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/arrow/catkin_ws/src/px4_fast_planner/models
```
Note that your home folder name can be different.

Also make sure that 
```sh
source $HOME/catkin_ws/devel/setup.bash
```
is before all the `export` lines.

Muzaib95Anjum, 2020-07-06:
@mzahana okay thanks i am re running it.

mzahana, 2020-07-06:
If your issue is solved please close it.

Muzaib95Anjum, 2020-07-06:
@mzahana i have re runned the file and after this command **roslaunch px4_fast_planner px4_fast_planner.launch** these errors are listed:

ERROR: cannot launch node of type [px4/px4]: Cannot locate node of type [px4] in package [px4]. Make sure file exists in package path and permission is set to executable (chmod +x)

[rvizvisualisation-16] process has died [pid 5438, exit code -6, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/muzaib/catkin_ws/src/px4_fast_planner/rviz/rviz.rviz __name:=rvizvisualisation __log:=/home/muzaib/.ros/log/8ac84420-bfb5-11ea-91fb-54ab3ae22b7e/rvizvisualisation-16.log].
log file: /home/muzaib/.ros/log/8ac84420-bfb5-11ea-91fb-54ab3ae22b7e/rvizvisualisation-16*.log

[gazebo-5] process has died [pid 5398, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/muzaib/catkin_ws/src/px4_fast_planner/worlds/outdoor_village.world __name:=gazebo __log:=/home/muzaib/.ros/log/8ac84420-bfb5-11ea-91fb-54ab3ae22b7e/gazebo-5.log].
log file: /home/muzaib/.ros/log/8ac84420-bfb5-11ea-91fb-54ab3ae22b7e/gazebo-5*.log

RVIZ opened and closed immediately and Gazebo has no models!!

mzahana, 2020-07-06:
How does your .bashrc look like?
Did you source it before you launch?
Also expect Gazebo to take some time the first time you run the setup as it downloads worrd models

Muzaib95Anjum, 2020-07-06:
My .bashrc 

source /opt/ros/melodic/setup.bash
source /home/muzaib/catkin_ws/devel/setup.bash
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/muzaib/catkin_ws/src/px4_fast_planner/models

Yes i sourced it when opened new terminal **source .bashrc**

mzahana, 2020-07-06:
you need to also have
```sh
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
```

Muzaib95Anjum, 2020-07-06:
This is my .bashrc

source /opt/ros/melodic/setup.bash
source /home/muzaib/catkin_ws/devel/setup.bash
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/muzaib/catkin_ws/src/px4_fast_planner/models

mzahana, 2020-07-06:
I am not sure why are you crossing 
source ~/Firmware/Tools/setup_gazebo.bash /Firmware /Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/Firmware/Tools/sitl_gazebo

are they not included in your bashrc?

Muzaib95Anjum, 2020-07-06:
@mzahana i am not crossing them when i paste them from terminal it gets crossed. Yes they are included in my .bashrc. And Gazebo is opened with the world but i cant see the drone in it. And rviz is not opened also kindly guide.

mzahana, 2020-07-06:
I am not sure what your issue is here. The setup seems correct. Are you able to launch PX$ SITL alone?

Muzaib95Anjum, 2020-07-06:
@ i have run the command **roslaunch px4_fast_planner px4_fast_planner.launch** it opened the following

- rqt_configure
- gazebo

But i cant see the drone in the Simulation and Rviz opened and immediately closed.. This is i am assking..

mzahana, 2020-07-06:
Can you run ?
```sh
cd ~/Firmware
make px4_sitl_default gazebo
```

Muzaib95Anjum, 2020-07-06:
@mzahana Yes i ran and gazebo opened with empty world and a Drone

mzahana, 2020-07-06:
That means PX4 SITL is compiled and successfully. Then, I see no reason why the setup should not work.

mzahana, 2020-07-06:
I just notice that your .bashrc is not correct. The `Firmware` folder should be in home i.e. `~/Firmware`, and the correct .bashrc should point to `~/Firmware` not `/Firmware`

Muzaib95Anjum, 2020-07-06:
@mzahana the project is started to work Fine now Thank you for your cooperation.

closed 2020-07-06 by mzahana

DavidLlantenC, 2021-07-26:
Hello! I am new to ROS and Linux. Right now I am doing research that requires the use of them.
When trying to run the ORB_SLAM2_CUDA project on a jetson nano, but I have an error specified as follows:

Resource not found: the following package was not found in <arg default = "$ find orb_slam2_cuda) /../../../ Vocabulary / ORBVoc.txt" name = "vocabularty_path" />: ORB_SLAM2_CUDA
ROS path [0] = / opt / ros / melodic / share / ros
ROS path [1] = / opt / ros / melodic / share /
The traceback for the exception was written to the log file

I don't know what to do to fix this error. I would appreciate your cooperation.

mzahana, 2021-07-26:
@DavidLlantenC Hi!

Unfortunately, this is not the right place for your question. It's better to ask on the corresponding repository issue tracker.

Tanmaywankhede07, 2021-10-19:
Hello !!..i'm trying to the simulation but it's show the same like @Muzaib95Anjum and my .bashrc looks like 
.
.

source /opt/ros/melodic/setup.bash
source /home/tanmay/catkin_ws/devel/setup.bash
export PATH=/usr/lib/ccache:$PATH
export GAZEBO_MODEL_PATH=~/ardupilot_gazebo/model
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/tanmay/catkin_ws/src/px4_fast_planner/models
.
.
.
**And the error shown in the terminal is** 
Resource not found: The following package was not found in <arg default="$(find mavlink_sitl_gazebo)/worlds/empty.world" name="world"/>: mavlink_sitl_gazebo
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/tanmay/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

Tanmaywankhede07, 2021-10-19:
And after adding /opt/ros/melodic/share to .bashrc, I'm getting this error 
.
.
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
started roslaunch server http://tanmay-Nitro-AN515-54:39111/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera_pose_publisher/child_frame: camera_link
 * /camera_pose_publisher/parent_frame: world
 * /camera_pose_publisher/pose_topic: camera/pose
 * /fast_planner_node/bspline/limit_acc: 2.5
 * /fast_planner_node/bspline/limit_ratio: 1.1
 * /fast_planner_node/bspline/limit_vel: 1.0
 * /fast_planner_node/fsm/act_map: False
 * /fast_planner_node/fsm/flight_type: 1
 * /fast_planner_node/fsm/thresh_no_replan: 2.0
 * /fast_planner_node/fsm/thresh_replan: 0.5
 * /fast_planner_node/fsm/waypoint0_x: 19.0
 * /fast_planner_node/fsm/waypoint0_y: 0.0
 * /fast_planner_node/fsm/waypoint0_z: 1.0
 * /fast_planner_node/fsm/waypoint1_x: 0.0
 * /fast_planner_node/fsm/waypoint1_y: 10.0
 * /fast_planner_node/fsm/waypoint1_z: 1.0
 * /fast_planner_node/fsm/waypoint2_x: -19.0
 * /fast_planner_node/fsm/waypoint2_y: 0.0
 * /fast_planner_node/fsm/waypoint2_z: 1.0
 * /fast_planner_node/fsm/waypoint_num: 2
 * /fast_planner_node/heading_planner/half_vert_num: 3
 * /fast_planner_node/heading_planner/lambda1: 2.0
 * /fast_planner_node/heading_planner/lambda2: 1.0
 * /fast_planner_node/heading_planner/max_yaw_rate: 1.04719753333
 * /fast_planner_node/heading_planner/w: 10.0
 * /fast_planner_node/heading_planner/yaw_diff: 0.0872664611111
 * /fast_planner_node/manager/clearance_threshold: 0.2
 * /fast_planner_node/manager/control_points_distance: 0.3
 * /fast_planner_node/manager/dynamic_environment: 0
 * /fast_planner_node/manager/local_segment_length: 7.0
 * /fast_planner_node/manager/max_acc: 2.5
 * /fast_planner_node/manager/max_jerk: 4.0
 * /fast_planner_node/manager/max_vel: 1.0
 * /fast_planner_node/manager/use_geometric_path: False
 * /fast_planner_node/manager/use_kinodynamic_path: False
 * /fast_planner_node/manager/use_optimization: True
 * /fast_planner_node/manager/use_topo_path: True
 * /fast_planner_node/optimization/algorithm1: 15
 * /fast_planner_node/optimization/algorithm2: 11
 * /fast_planner_node/optimization/dist0: 0.4
 * /fast_planner_node/optimization/dist1: 0.0
 * /fast_planner_node/optimization/lambda1: 10.0
 * /fast_planner_node/optimization/lambda2: 5.0
 * /fast_planner_node/optimization/lambda3: 0.0
 * /fast_planner_node/optimization/lambda4: 0.001
 * /fast_planner_node/optimization/lambda5: 1.5
 * /fast_planner_node/optimization/lambda6: 10.0
 * /fast_planner_node/optimization/lambda7: 20.0
 * /fast_planner_node/optimization/max_acc: 2.5
 * /fast_planner_node/optimization/max_iteration_num1: 2
 * /fast_planner_node/optimization/max_iteration_num2: 300
 * /fast_planner_node/optimization/max_iteration_time1: 0.0001
[... 314 lines cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_reconfigure/param_plugin.py", line 35, in <module>
    from rqt_gui_py.plugin import Plugin
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui_py/plugin.py", line 33, in <module>
    from qt_gui.plugin import Plugin as Base
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin.py", line 31, in <module>
    from python_qt_binding.QtCore import QObject
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 276, in <module>
    getattr(sys, 'SELECT_QT_BINDING_ORDER', None),
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 155, in _load_pyqt
    _named_optional_import('PyQt5.%s' % module_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 142, in _named_optional_import
    _named_import(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
KeyboardInterrupt
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
@mzahana

abdouakhad, 2022-07-19:
Well in my case I just rebooted the computer with `sudo reboot now` after installing px4 dependencies and everything works finally

AlexWUrobot, 2022-10-20:
I have the same issue after install zsh terminal. I solve it by delete the build and devel folder and then catkin build again. After that, I close the terminal and reopen the the terminal. Then. it works