mzahana/px4_fast_planner #5: Set altitude when switching offboard mode

lgh5054, 2020-07-07:
Hello 
It automatically switches from simulation to offboard mode and starts with an altitude of 2m.

Can I change the starting altitude to 1m?

And can the avoidance radius be changed for larger drone frame use?

mzahana, 2020-07-07:
@lgh5054 Currently, initial target position is hard-coded in [here](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/master/geometric_controller/src/geometric_controller.cpp#L57). I will look into parameterizing it. For the avoidance radius will get back to you later on that.

Jaeyoung-Lim, 2020-07-08:
@lgh5054 This was fixed in https://github.com/Jaeyoung-Lim/mavros_controllers/pull/136 by @mzahana

mzahana, 2020-07-08:
@lgh5054 
This should be fixed by now. Pull latest master of `mavros_controllers` and `px4_fast_planner`. Then, you can modify the initial target position in [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L23-L25)

lgh5054, 2020-07-08:
@Jaeyoung-Lim @mzahana  thanks 
I'll report the results after the test.

lgh5054, 2020-07-15:
@mzahana I'm testing it on Realdrone.
When the arm and offboard are run using Qground control  the drone cannot be positioned and flies away.
Do I need to run arm and offboard using geometric_controller.launch?
And I use altitude using lidar.
But it seems to keep using baro here.
Can I use a lidar?

mzahana, 2020-07-16:
There could be many reasons for your issues.
* First make sure that you have stable odometry system, it doesn't have to be based on GPS
* Make sure you have correct [camera pose topic](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L16) published. That also requires correct transformations including [depth camera static TF to base_link](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L29). The camera pose topic is published by [this node](https://github.com/mzahana/px4_fast_planner/blob/master/scripts/camera_pose_publisher.py) which listens to TF from map frame to camera frame. So it's important to have the transformations correct. You can use RViz to help you visualize and debug the TFs
* The geometric controller has an initial position of (0,0,2). Depending on your odometry system, this might not be a good takeoff position for you.
* Make sure that there are setpoints published at `/mavros/setpoint_raw/attitude`. Otherwise the `OFFBOARD` mode will not be activated and the vehicle can go to failsafe mode which may result into a climb up to predefined failsafe altitude and then landing
* Make sure that MAVROS state looks as follows (mainly that `OFFBOARD` mode is activated)
```sh
header: 
  seq: 379
  stamp: 
    secs: 3689
    nsecs: 908000000
  frame_id: ''
connected: True
armed: True
guided: True
manual_input: False
mode: "OFFBOARD"
system_status: 4
```
Hopefully this helps.

mzahana, 2020-07-16:
For the LIDAR to take effect, you need to set the right parameters in QGC to use it and fuse it in the EKF2

lgh5054, 2020-07-16:
@mzahana thanks resolved

closed 2020-07-16 by lgh5054

mzahana, 2020-07-16:
@lgh5054 great. It would be nice if you share how you solved your issue so others can benefit from it in case they encounter the same problem. Thanks.

lgh5054, 2020-07-17:
@mzahana That's a good idea, but...
It was not a px4_fast_planner problem
mavros_controllers needs to be modified!

Jaeyoung-Lim, 2020-07-17:
@lgh5054 Could you explain what needed to be modified?

lgh5054, 2020-07-17:
@Jaeyoung-Lim The geometric_controller thrust has been adjusted because it uses a racing motor. So it works normally. 

However, when rotating the YAW axis, the drone rotates so fast that it loses direction. Can you reduce the speed of YAW rotation?

lgh5054, 2020-07-17:
@mzahana lift the limit
<arg name="map_size_x" default="40.0"/>
<arg name="map_size_y" default="40.0"/>
<arg name="map_size_z" default=" 5.0"/>

What do I do?

Isn't Kino planner supported?

mzahana, 2020-07-17:
@lgh5054 I am sorry, I don't understand your question. kino_planner is supported. You need to set `use_kino_planner` to `true`
https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L10

Please open another issue if it's not related to this issue(Set altitude when switching offboard mode). Thanks.

atmaca11, 2021-05-06:
@Jaeyoung-Lim What components does your quad have? I want to make this project in real quadcopter. Can you share your setup ?