mzahana/px4_fast_planner #8: move_base z axis goal position

yuthikasagarage, 2020-07-21:
when giving a xy goal in move_base in rviz,  vertical distance to the drone is low, sometimes the drone touches the ground. so when we give a goal in rviz as a 2d goal, does it assume 0 for the vertical axis?

mzahana, 2020-07-21:
In the original implemetnation of Fast-Planner the height was fixed to 1m to avoid arbitrary height sent by Rviz 2D goal (it could be even negative!). I modified this to accept user defined height, and it is not recommended to use Rviz 2D goal in this case as it could be any height. The setup script i included builds installs the modified version. to control what heigh you want to reach, you need to publish to the move_base goal the required  specific value.

yuthikasagarage, 2020-07-21:
got it. thanks for the clarification. closing.

closed 2020-07-21 by yuthikasagarage