mzahana/px4_fast_planner #9: Change Drone Start Position

lgh5054, 2020-07-23:
Hello

I'd like to start take off the drone at x = -40

<arg name="init_force_x" default="0.0" />
If you change the <arg name="init_force_x" default="-40.0" />, it will fly in the -40 direction after takeoff.

After Launch File

Can a drone start at x= -40?

mzahana, 2020-08-01:
@lgh5054 Please check the updated launch file (https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L36-L37). Now you can set the initial position of the drone in Gazebo. Please note that this is different from the position estimate provided by PX4 estimator. Although you can arbitrary change the initial position in Gazebo, this is still going to be the zero position of the PX4 local frame.

lgh5054, 2020-08-03:
I appreciate you responding

I can set the initial position of the drone in Gazebo

but I need is  set the PX4 local frame .... What do I do?

mzahana, 2020-08-03:
I don't think there is a direct way to do that.

lgh5054, 2020-08-03:
@mzahana  thanks mzahana

closed 2020-08-03 by lgh5054

mzahana, 2020-08-03:
One, indirect, way to do it is to write a node that publishes vision pose estimates based on Gazebo pose of the drone model. This will be passed to MAVROS as vision pose estimate (https://dev.px4.io/v1.10/en/ros/external_position_estimation.html). You will also need to set the `EKF2_AID_MASK` in the corresponding PX4 startup file appropriately to accept vision estimates instead of GPS. Additionally, you will need to set `EKF2_HGT_MODE` to use height estimate from vision. For further information, please refer to the related documentation.