mzahana/px4_fast_planner #10: ERROR: cannot launch node of type [plan_manage/fast_planner_node]: Cannot locate node of type [fast_planner_node] in package [plan_manage] ,similar issue for nodes [traj_server] and  [geometric_controller_node]

Akshit-Patel, 2020-07-30:
Hi,
This is awesome and very exciting. I am new to this community and was working on an autonomous indoor drone project where I am aiming drone to cross some frames in the aisle and along with that avoid obstacles in between. I am looking forward to this project for some help. I have been trying to set up this repository for the past few days and haven't succeeded till now.

Version - PX4 Firmware v1.10.1, Ubuntu 18.04, ROS melodic

1.) I initially set up my ROS workspace in a directory other than home due to some space issues
As mentioned in instructions I cloned px4_fast_planner into src directory of my workspace i.e., `/mnt/Softwares/ROS/Drone/src`,  I installed all dependencies through `setup.sh` which created a new workspace in the home directory and was automatically sourced into bashrc by `setup.sh`

So I modified my bashrc (i.e., explicitly exported `px4_fast_planner` and `px4_fast_planner/models` into ROS_PACKAGE_PATH and GAZEBO_MODEL_PATH) as shown below to meet my requirements. The one commented out is my old workspace.

```
source /opt/ros/melodic/setup.bash
# source /mnt/Softwares/ROS/Drone/devel/setup.bash
source /home/akshit/catkin_ws/devel/setup.bash
source /mnt/Softwares/ROS/Drone/Firmware/Tools/setup_gazebo.bash /mnt/Softwares/ROS/Drone/Firmware/ /mnt/Softwares/ROS/Drone/Firmware/build/px4_sitl_default
export PATH="$HOME/.yarn/bin:$HOME/.config/yarn/global/node_modules/.bin:$PATH"
export PYENV_ROOT="/mnt/Softwares/soft_installed_ubuntu/pyenv/pyenv"
export PATH="$PYENV_ROOT/bin:$PATH"
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/mnt/Softwares/ROS/Drone/Firmware:/mnt/Softwares/ROS/Drone/Firmware/Tools/sitl_gazebo:/mnt/Softwares/ROS/Drone/src/px4_fast_planner
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/mnt/Softwares/ROS/Drone/GAAS/simulator/models:/mnt/Softwares/ROS/practice/world:/mnt/Softwares/ROS/Drone/Firmware/Tools/sitl_gazebo/models:/mnt/Softwares/ROS/Drone/gazebo-model/gazebo_models:/mnt/Softwares/ROS/Drone/src/avoidance/avoidance/sim/models:/mnt/Softwares/ROS/Drone/src/avoidance/avoidance/sim/worlds:/mnt/Softwares/ROS/Drone/src/px4_fast_planner/models
```
then I tried launching px4_fast_planner.launch and got this error (log1 file in comments)
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
process[gazebo-4]: started with pid [17065]
Error: Unknown model 'iris_depth_camera'
ERROR [px4] Startup script returned with return value: 256
process[gazebo_gui-5]: started with pid [17073]
process[vehicle_spawn_akshit_Inspiron_7577_17017_1700258111149677753-6]: started with pid [17078]
================================================================================REQUIRED process [sitl-3] has died!
process has died [pid 17051, exit code 255, cmd /mnt/Softwares/ROS/Drone/Firmware/build/px4_sitl_default/bin/px4 /mnt/Softwares/ROS/Drone/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS __name:=sitl __log:=/home/akshit/.ros/log/61a6707e-d1e1-11ea-be79-34415d6d1c2b/sitl-3.log].
log file: /home/akshit/.ros/log/61a6707e-d1e1-11ea-be79-34415d6d1c2b/sitl-3*.log
Initiating shutdown!
========================
```
to solve this I created a copy of 10016_iris in my `/mnt/Softwares/ROS/Drone/Firmware/ROMFS/px4fmu_common/init.d-posix` directory and renamed it to `10017_iris_depth_camera` to create a uniqueID for this model which was missing there. I am not sure whether it's correct to do this but it solved my error.

On trying again I was successfully able to launch the gazebo world and view sensor depth data in rviz. But still, I am not able to launch my drone either through QgroundControl or by publishing in topic  `/move_base_simple/goal `.I faced these errors. (log2 file in comment)
```
ERROR: cannot launch node of type [plan_manage/fast_planner_node]: Cannot locate node of type [fast_planner_node] in package [plan_manage]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [plan_manage/traj_server]: Cannot locate node of type [traj_server] in package [plan_manage]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[waypoint_generator-11]: started with pid [6318]
ERROR: cannot launch node of type [geometric_controller/geometric_controller_node]: Cannot locate node of type [geometric_controller_node] in package [geometric_controller]. Make sure file exists in package path and permission is set to executable (chmod +x)
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560

[ERROR] [1596116398.936015, 3491.900000]: Spawn service failed. Exiting.
```
I tried doing `sudo chmod +x -R catkin_ws` ,but it still shows the same error.

2.) Can I use this to go through narrow spaces like frames(reference image attached in comments, landing gears can be neglected). If not possible directly is there any way i can turn off px4_fast_planner and continue with some other code when I am near to the frame and after that activate this planner again.

Thanks

Akshit-Patel, 2020-07-30:
Relevant files 

[log1.txt](https://github.com/mzahana/px4_fast_planner/files/5001888/log1.txt)
[log2.txt](https://github.com/mzahana/px4_fast_planner/files/5001889/log2.txt)
[drone-image](https://user-images.githubusercontent.com/29256293/88936865-01ccfb80-d2a1-11ea-91c6-17e96b8d43a9.png)

Akshit-Patel, 2020-07-31:
I changed the location of px4_fast_planner to the new workspace and modified .bashrc file accordingly. Again rviz and gazebo popped up with same errors as mentioned earlier in final step 
  
I think the above issue is mostly due to me,I might have ignored build errors earlier by mistake, which I just realised after running `setup.sh `or `catkin build` again. There were many point cloud errors as shown below

[Screenshot from 2020-08-01 04-09-08](https://user-images.githubusercontent.com/29256293/89083211-dedc3d80-d3ad-11ea-915a-9ce1d0aaf464.png)

[Screenshot from 2020-08-01 04-09-32](https://user-images.githubusercontent.com/29256293/89083230-fb787580-d3ad-11ea-8143-e4e209e7ece5.png)

[Screenshot from 2020-08-01 04-09-20](https://user-images.githubusercontent.com/29256293/89083233-00d5c000-d3ae-11ea-86a7-076327d34fad.png)

Could this be due to PCL version differences? I am using pcl-1.11

Thanks

mzahana, 2020-08-01:
Hi @Akshit-Patel . Thanks for using `px4_fast_planner`. It was tested with `pcl-1.8`. I am not sure if it works with `1.11`

Akshit-Patel, 2020-08-01:
Thanks for your reply, 
I compiled pcl-1.11 from source and pcl-1.8 was already installed with `apt-get` which was leading to build issues. After removing pcl-1.11 I was successfully able to build this package and launch nodes properly. But these issue stil remain 

```
[INFO] [1596279718.602390, 0.000000]: Calling service /gazebo/spawn_sdf_model
[INFO] [1596279718.728123, 3491.900000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name iris_depth_camera
[ERROR] [1596279718.729415, 3491.900000]: Spawn service failed. Exiting.
Warning [parser.cc:950] XML Element[vertical_fov], child of element[camera] not defined in SDF. Ignoring[vertical_fov]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[vehicle_spawn_akshit_Inspiron_7577_13249_5751270001766641971-7] process has died [pid 13329, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /home/akshit/catkin_ws/src/px4_fast_planner/models/iris_depth_camera/iris_depth_camera.sdf -model iris_depth_camera -x 0 -y 0 -z 0.2 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_akshit_Inspiron_7577_13249_5751270001766641971
```
I don't know what is the issue exactly but I think it might be related to `10017_iris_depth_camera`, the file which I created in `/Firmware/ROMFS/px4fmu_common/init.d-posix` for eliminating startup script error. Currently, I have simply defined it to just contain one line, `sh /etc/init.d-posix/10016_iris`

Also, there were these errors

```
INFO  [ecl/EKF] 3192000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 3192000: reset position to last known position
INFO  [ecl/EKF] 3192000: reset velocity to zero
WARN  [mc_pos_control] Offboard activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed
[FSM]: from INIT to WAIT_TARGET
[ INFO] [1596286073.542292840, 3495.300000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1596286073.670281619, 3495.428000000]: Home pose initialized to: position: 
  x: 0
  y: 0
  z: -0.00843018
orientation: 
  x: 0.0097147
  y: -0.0111899
  z: 0.0228288
  w: -0.99963

[ INFO] [1596286074.171937837, 3495.928000000]: Got pose! Drone Ready to be armed.
state: WAIT_TARGET
state: WAIT_TARGET
INFO  [ecl/EKF] 5284000: reset position to GPS
INFO  [ecl/EKF] 5284000: reset velocity to GPS
INFO  [ecl/EKF] 5284000: starting GPS fusion
state: WAIT_TARGET
[rqt_reconfigure-13] process has finished cleanly
log file: /home/akshit/.ros/log/3240d9bc-d3f5-11ea-9145-34415d6d1c2b/rqt_reconfigure-13*.log
state: WAIT_TARGET
[rvizvisualisation-16] process has finished cleanly
log file: /home/akshit/.ros/log/3240d9bc-d3f5-11ea-9145-34415d6d1c2b/rvizvisualisation-16*.log
state: WAIT_TARGET
state: WAIT_TARGET
state: WAIT_TARGET
state: WAIT_TARGET
state: WAIT_TARGET
state: WAIT_TARGET
WARN  [PreFlightCheck] Preflight Fail: Gyro #0 uncalibrated
WARN  [PreFlightCheck] Primary gyro not found
WARN  [PreFlightCheck] system power unavailable
[ERROR] [1596286083.694041478, 3505.436000000]: FCU: Preflight Fail: Gyro #0 uncalibrated
[ERROR] [1596286083.744923992, 3505.488000000]: FCU: Primary gyro not found
[ERROR] [1596286083.793577566, 3505.536000000]: FCU: system power unavailable
state: WAIT_TARGET
```
I tried changing `EKF2_AID_MASK` parameter to 1 and 24 through QGC and than rebooting the vehicle but could not eliminate the gyro error.

mzahana, 2020-08-02:
The install script already copies the required PX4 startup file to the required folder inside PX4 Firmware folder. A copy is available here (https://github.com/mzahana/px4_fast_planner/blob/master/config/10017_iris_depth_camera). Also the install script adds the necessary model path to the gazebo model path (https://github.com/mzahana/px4_fast_planner/blob/master/install/setup.sh#L138). This is tested with PX4 v1.10.1. Otherwise, I am not sure what the problem is. Also I am assuming that you are using the iris model with depth camera in this package. If you are using your own custom model, you will have to add it to the path and have its own PX4 startup file added properly too.

Akshit-Patel, 2020-08-13:
Thanks,
I was using the iris model with the depth camera package and faced those issues. Finally, I tried to set it up from start. I was again required to create a file  `/Firmware/ROMFS/px4fmu_common/init.d-posix/10017_iris_depth_camera`. Now everything seems to work fine, vehicle launches properly, set points can be published, and it reaches the set point by avoiding obstacles. Although this error still shows up,

```
[ERROR] [1597360258.058553, 3491.900000]: Spawn service failed. Exiting.
Warning [parser.cc:950] XML Element[vertical_fov], child of element[camera] not defined in SDF. Ignoring[vertical_fov]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[ INFO] [1597360258.663421458, 3491.900000000]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1597360258.665692584, 3491.900000000]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[vehicle_spawn_akshit_Inspiron_7577_5021_1370868167067778571-7] process has died [pid 5924, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /home/akshit/catkin_ws/src/px4_fast_planner/models/iris_depth_camera/iris_depth_camera.sdf -model iris_depth_camera -x 0 -y 0 -z 0.2 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_akshit_Inspiron_7577_5021_1370868167067778571 __log:=/home/akshit/.ros/log/215fff6e-ddba-11ea-ae23-34415d6d1c2b/vehicle_spawn_akshit_Inspiron_7577_5021_1370868167067778571-7.log].
log file: /home/akshit/.ros/log/215fff6e-ddba-11ea-ae23-34415d6d1c2b/vehicle_spawn_akshit_Inspiron_7577_5021_1370868167067778571-7*.log
INFO  [simulator] Simulator connected on TCP port 4560.
```
Log File 
[vehicle_spawn_akshit_Inspiron_7577_5021_1370868167067778571-7.log](https://github.com/mzahana/px4_fast_planner/files/5071910/vehicle_spawn_akshit_Inspiron_7577_5021_1370868167067778571-7.log)

Error Screenshot
[error](https://user-images.githubusercontent.com/29256293/90196471-7b103680-dde9-11ea-9891-cf307b2343f9.png)

 <p align="center">
  [image]
 </p>

Not sure whether this can be closed.

mzahana, 2020-08-17:
Good that you got it working. Since this error is not related to this package (this is PX4-related), I think this issue can be closed.

closed 2020-08-17 by Akshit-Patel

hjl0404, 2021-12-17:
Hello, I have the same problem as you (below), and the version selection is consistent with your version.
How did you solve this problem successfully? Besides deleting Pcl-1.11, what else did you do? Because I only have Pcl-1.8 and did not find Pcl-1.11.
I have been troubled by this problem for a long time. I am looking forward to receiving your reply. Thank you.

“cannot launch node of type [plan_manage/fast_planner_node]: Cannot locate node of type [fast_planner_node] in package [plan_manage] ,similar issue for nodes [traj_server] and [geometric_controller_node]”