mzahana/px4_fast_planner #11: ERROR: cannot launch node of type [px4/px4]: Cannot locate node of type [px4] in package [px4]. Make sure file exists in package path and permission is set to executable (chmod +x)

ziniu888, 2020-08-11:
Hi,

It seems that px4 node cannot be launched.

```
ERROR: cannot launch node of type [px4/px4]: Cannot locate node of type [px4] in package [px4]. Make sure file exists in package path and permission is set to executable (chmod +x)
```

However, px4 simulation by
```
make px4_sitl_default gazebo
```
is working correctly. Could anyone help me with this issue?

mzahana, 2020-08-11:
@ziniu888 Have you tried
```sh
cd ~
source .bashrc
```
?

ziniu888, 2020-08-11:
@mzahana Sure. I followed this #4 issue and made some changes. 

```

source ~/Documents/mza_px4_fast_plan_ws/devel/setup.bash
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/ziniu/Documents/mza_px4_fast_plan/src/px4_fast_planner/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/.gazebo/models
```

ziniu888, 2020-08-11:
Here is my log.

[log_7f2b74.zip](https://github.com/mzahana/px4_fast_planner/files/5056431/log_7f2b74.zip)

ziniu888, 2020-08-11:
The issue above may caused by user`s permission, Now I have a solution:

- Start a new terminal
```
no_sim=1 make px4_sitl_default gazebo
```
Then px4 node work correctly.

closed 2020-08-11 by ziniu888