mzahana/px4_fast_planner #12: px4_fast_planner.launch fails to Start

mjs513, 2020-10-04:
By way of warning really new to ROS, linux and the rest and am trying to get this to work.  Anyway, when I do `roslaunch px4_fast_planner px4_fast_planner.launch`   it goes through starting but then exits due to errors:
```

roslaunch px4_fast_planner px4_fast_planner.launch
... logging to /home/ubuntu/.ros/log/9e7b489c-0668-11eb-8cf3-dca632bd9293/roslaunch-ubuntu-4619.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
started roslaunch server http://ubuntu:40055/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera_pose_publisher/child_frame: camera_link
 * /camera_pose_publisher/parent_frame: world
 * /camera_pose_publisher/pose_topic: camera/pose
 * /fast_planner_node/bspline/limit_acc: 2.5
 * /fast_planner_node/bspline/limit_ratio: 1.1
 * /fast_planner_node/bspline/limit_vel: 1.0
 * /fast_planner_node/fsm/act_map: False
 * /fast_planner_node/fsm/flight_type: 1
 * /fast_planner_node/fsm/thresh_no_replan: 2.0
 * /fast_planner_node/fsm/thresh_replan: 0.5
 * /fast_planner_node/fsm/waypoint0_x: 19.0
 * /fast_planner_node/fsm/waypoint0_y: 0.0
 * /fast_planner_node/fsm/waypoint0_z: 1.0
 * /fast_planner_node/fsm/waypoint1_x: 0.0
 * /fast_planner_node/fsm/waypoint1_y: 10.0
 * /fast_planner_node/fsm/waypoint1_z: 1.0
 * /fast_planner_node/fsm/waypoint2_x: -19.0
 * /fast_planner_node/fsm/waypoint2_y: 0.0
 * /fast_planner_node/fsm/waypoint2_z: 1.0
 * /fast_planner_node/fsm/waypoint_num: 2
 * /fast_planner_node/heading_planner/half_vert_num: 3
 * /fast_planner_node/heading_planner/lambda1: 2.0
 * /fast_planner_node/heading_planner/lambda2: 1.0
 * /fast_planner_node/heading_planner/max_yaw_rate: 1.04719753333
 * /fast_planner_node/heading_planner/w: 10.0
 * /fast_planner_node/heading_planner/yaw_diff: 0.0872664611111
 * /fast_planner_node/manager/clearance_threshold: 0.2
 * /fast_planner_node/manager/control_points_distance: 0.3
 * /fast_planner_node/manager/dynamic_environment: 0
 * /fast_planner_node/manager/local_segment_length: 7.0
 * /fast_planner_node/manager/max_acc: 2.5
 * /fast_planner_node/manager/max_jerk: 4.0
 * /fast_planner_node/manager/max_vel: 1.0
 * /fast_planner_node/manager/use_geometric_path: False
 * /fast_planner_node/manager/use_kinodynamic_path: False
 * /fast_planner_node/manager/use_optimization: True
 * /fast_planner_node/manager/use_topo_path: True
 * /fast_planner_node/optimization/algorithm1: 15
 * /fast_planner_node/optimization/algorithm2: 11
 * /fast_planner_node/optimization/dist0: 0.4
 * /fast_planner_node/optimization/dist1: 0.0
 * /fast_planner_node/optimization/lambda1: 10.0
 * /fast_planner_node/optimization/lambda2: 5.0
 * /fast_planner_node/optimization/lambda3: 0.0
 * /fast_planner_node/optimization/lambda4: 0.001
 * /fast_planner_node/optimization/lambda5: 1.5
[... 506 lines cut ...]
  File "/usr/lib/python2.7/sre_compile.py", line 132, in _compile
    _compile(code, av[2], flags)
  File "/usr/lib/python2.7/sre_compile.py", line 107, in _compile
    _compile_charset(av, flags, code, fixup, fixes)
  File "/usr/lib/python2.7/sre_compile.py", line 233, in _compile_charset
    emit(OPCODES[op])
KeyboardInterrupt
[gazebo-5] killing on exit
[sitl-4] killing on exit
ERROR [px4] Startup script returned with return value: 2
pxh> [tf_camera-3] killing on exit
[world_map_linker-2] killing on exit
des manager
[sitl-4] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
Any help would be appreciated.

Thanks
MIke

mjs513, 2020-10-04:
This is the resuts of cmake:
```
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-my1vz4nl/pandas/

ubuntu@ubuntu:~/catkin_ws/src/px4_fast_planner/install$ ./setup.sh
 Do you want to build PX4 v1.10.1? (y) for simulation (n) if setting this up on on-barod computer 
y
 PX4 v1.10.1 will be built 
./setup.sh: line 31: catkin: command not found
./setup.sh: line 32: catkin: command not found
./setup.sh: line 33: catkin: command not found
source /home/ubuntu/catkin_ws/devel/setup.bash
 Setting up Px4 v1.10.1 
Reading package lists...
Building dependency tree...
Reading state information...
lsb-core is already the newest version (9.20170808ubuntu1).
ca-certificates is already the newest version (20190110~18.04.1).
gnupg is already the newest version (2.2.4-1ubuntu1.3).
wget is already the newest version (1.19.4-1ubuntu2.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installing PX4 general dependencies
Hit:1 http://packages.ros.org/ros/ubuntu bionic InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.4ubuntu1).
ccache is already the newest version (3.4.1-1).
doxygen is already the newest version (1.8.13-10).
make is already the newest version (4.1-9.1ubuntu1).
python3-setuptools is already the newest version (39.0.1-2).
unzip is already the newest version (6.0-21ubuntu1).
zip is already the newest version (3.0-11build1).
astyle is already the newest version (3.1-1ubuntu2).
cppcheck is already the newest version (1.82-1).
lcov is already the newest version (1.13-3).
ninja-build is already the newest version (1.8.2-1).
python3-wheel is already the newest version (0.30.0-0.2).
shellcheck is already the newest version (0.4.6-1).
cmake is already the newest version (3.10.2-1ubuntu2.18.04.1).
file is already the newest version (1:5.32-2ubuntu0.4).
g++ is already the newest version (4:7.4.0-1ubuntu2.3).
gcc is already the newest version (4:7.4.0-1ubuntu2.3).
gdb is already the newest version (8.1-0ubuntu3.2).
git is already the newest version (1:2.17.1-1ubuntu0.7).
python3 is already the newest version (3.6.7-1~18.04).
python3-dev is already the newest version (3.6.7-1~18.04).
rsync is already the newest version (3.1.2-2.1ubuntu1.1).
xsltproc is already the newest version (1.1.29-5ubuntu0.2).
clang is already the newest version (1:6.0-41~exp5~ubuntu1).
clang-tidy is already the newest version (1:6.0-41~exp5~ubuntu1).
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Installing PX4 Python3 dependencies
Collecting argparse>=1.2 (from -r /home/ubuntu/catkin_ws/src/px4_fast_planner/install/px4_requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/f2/94/3af39d34be01a24a6e65433d19e107099374224905f1e0cc6bbe1fd22a2f/argparse-1.4.0-py2.py3-none-any.whl
Collecting cerberus (from -r /home/ubuntu/catkin_ws/src/px4_fast_planner/install/px4_requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/90/a7/71c6ed2d46a81065e68c007ac63378b96fa54c7bb614d653c68232f9c50c/Cerberus-1.3.2.tar.gz
Collecting empy>=3.3 (from -r /home/ubuntu/catkin_ws/src/px4_fast_planner/install/px4_requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/3b/95/88ed47cb7da88569a78b7d6fb9420298df7e99997810c844a924d96d3c08/empy-3.3.4.tar.gz
Collecting jinja2>=2.8 (from -r /home/ubuntu/catkin_ws/src/px4_fast_planner/install/px4_requirements.txt (line 4))
  Using cached https://files.pythonhosted.org/packages/30/9e/f663a2aa66a09d838042ae1a2c5659828bb9b41ea3a6efa20a20fd92b121/Jinja2-2.11.2-py2.py3-none-any.whl
Collecting matplotlib (from -r /home/ubuntu/catkin_ws/src/px4_fast_planner/install/px4_requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/2b/4c/fe4b36325795524f35d39edc390c89584e9a901df9e615df6f5effddaa0e/matplotlib-3.3.2.tar.gz
[... 1105 lines cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui_py/plugin.py", line 33, in <module>
    from qt_gui.plugin import Plugin as Base
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin.py", line 31, in <module>
    from python_qt_binding.QtCore import QObject
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 276, in <module>
    getattr(sys, 'SELECT_QT_BINDING_ORDER', None),
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 155, in _load_pyqt
    _named_optional_import('PyQt5.%s' % module_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 142, in _named_optional_import
    _named_import(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
KeyboardInterrupt
[sitl-4] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
ubuntu@ubuntu:~/catkin_ws/src/px4_fast_planner/install$ 
```

mjs513, 2020-10-04:
Going to close this - found a solution here:
https://community.emlid.com/t/mavros-start-error-egm96-5-pgm-not-readable/15242

sudo /opt/ros/kinetic/lib/mavros/install_geographiclib_datasets.sh

mjs513, 2020-10-04:
Got it to load and run but when i try to publish the move command per the readme nothing seems to happen,  This is what i am seeing on the screen is it right>
[image]

mzahana, 2020-10-09:
Hi @mjs513 
I believe more information is needed in order to debug your problem. What is the list of running nodes?
Did the drone go to OFFBOARD mode? you can check the `mavros/state` topic for that.

mjs513, 2020-10-10:
Hi @mzahana 
Thanks for getting back to me - sorry it took me so long to respond. 

I started from scratch this morning and looking at the output of the build process more carefully and figuring a few thing out like i didn't have something installed properly I rebuilt the package.  After doing that I ran the test case again and this time everything worked as advertised.

On the PI4 it is rather slow though, is there a way to use a default world or simpler world?

going to close this right now.

closed 2020-10-10 by mjs513