mzahana/px4_fast_planner #16: Experiment in real world

codespring-lh, 2021-04-22:
Hi,I want to do some experiment using Fast-Planner in real world,but I have met some problems.I used Vins-Fusion with D435i and I have modified kino_replan.lauch.But there is no map.
Hoping for your reply.
[image]
[image]
[image]
[image]

codespring-lh, 2021-04-22:
And here is my tf_tree.
[image]

mzahana, 2021-10-15:
Refer to #17

closed 2021-10-15 by mzahana