mzahana/px4_fast_planner #18: Gazebo Simulation Environment

Kucukcollu, 2021-05-16:
Hi, I am working on this project and I am trying to change the simulation environment with different one. Unfortunately, I could not change the simulation environment. I created a new world with ".world" extension file but it dis not work. In addition of that I tried to remove current objects but the ground did not remove. Can you help me about where part of the project should I change?

beomsu7, 2021-05-17:
the .world file is in
px4_fast_planner/worlds/

and now the world is written in
px4_fast_planner/launch/mavros.launch line 34 world arg

so you can edit the file 'px4_fast_planner/worlds/outdoor_village.world'
or add new world file and edit the line 34 of 'px4_fast_planner/launch/mavros.launch'

Kucukcollu, 2021-05-17:
Firstly, thank you for your reply but it was not the case that I faced up. I tried to change the world but the drone did not flt. Then, I edited the file with gedit and it works for me. Thanks!

closed 2021-05-20 by Kucukcollu