mzahana/px4_fast_planner #24: The px4 shakes left and right when taking off.

one-DS, 2021-09-27:
When applied to the physical aircraft, the UAV shakes left and right when taking off. How to adjust it? We look forward to your reply. Thank you!

mzahana, 2021-09-27:
@one-DS you may need to tune the geometric controller.

one-DS, 2021-10-01:
@mzahana  Hello, thank you for taking the time to reply to me. geometric_ The parameter range of the controller is 0.1 ~ 20. By adjusting the parameters, the UAV does not shake as much as before, but the parameters have been adjusted to the end, and the situation is still not ideal. Therefore, I want to ask you, is this method feasible by modifying the controller program to expand the parameter range and make a larger range adjustable? thank you!

mzahana, 2021-10-15:
@one-DS you may change them in the launch file, [here](https://github.com/mzahana/px4_fast_planner/blob/master/launch/px4_fast_planner.launch#L62-L75)

Goldsaints, 2021-12-13:
> @mzahana Thank you! The problem has been solved,but I want to know how these parameters control the attitude of UAV, like kpos and kvel. I don't know what these two mean at present.

Hello! I have the same problem. Can I know your parameter value?

daishuaibi, 2022-01-07:
@one-DS Hello, I have met the same problem, may I ask how to set  parameters? I have tried many times but it failed

Goldsaints, 2022-01-07:
Hi, I reinstalled the mavros_controllers. I used this version, you can look this photo. I hope it works for you. 
https://github.com/Jaeyoung-Lim/mavros_controllers/releases

daishuaibi, 2022-01-07:
@Goldsaints Thank you very much for your reply. I would like to ask if this mavros_controllers is to replace the original file and run .setup.sh to reinstall,I tried but it didn't work，Is there something wrong with my procedure？

ZhangXiaolei314, 2022-03-20:
> @Goldsaints Thank you very much for your reply. I would like to ask if this mavros_controllers is to replace the original file and run .setup.sh to reinstall,I tried but it didn't work，Is there something wrong with my procedure？

you should change the file px4_fast_planner/launch/geometric_controller.launch like this 
[image]
these three lines are necessary 
[image]