mzahana/px4_fast_planner #25: Altitude drops

eldewany, 2021-09-30:
Hello, i successfully setup the simulation environment and every thing runs smoothly, the issue is that after the drone hovers I publish the message at a certain altitude (tried the same of the hovering and different one) and in both cases the drone altitude drops down to 0. What might the problem be ?

mzahana, 2021-09-30:
@eldewany please share detailed steps.

eldewany, 2021-09-30:
@mzahana I edited the launch file to change the takeoff altitude, velocity limits and what not. Launch using px4_fast_planner.Launch file, everything starts up, all nodes are running with no errors, rviz is working. I then publish a message on the move_base topic as described using rqt publisher, it starts to drop in altitude but tries to move in the direction specified. Btw I'm on Ubuntu 18.04, melodic, px4 1.11.3

one-DS, 2021-10-01:
@eldewany  Did you not release altitude Z but only x and y when you released waypoint information?

eldewany, 2021-10-03:
@one-DS yes i did, here is a screen shot of the console when triggered
[image]

mzahana, 2021-10-03:
@eldewany try to publish only one message when you wish to change the goal point, not at a specific rate as you did.
Also, keep in mind that you are constrained by the map size. If you are using the default map size (40m) in x/y axes,  then your goal x position is actually out of bounds, and it won't work as expected.

eldewany, 2021-10-03:
@mzahana I did as you said, (Changed, x and y to be < 40 ) and published the message once and it still falls, although when it reaches the goal (while crawling on the ground) it takes off again to the specified altitude

mzahana, 2021-10-03:
I am not sure what's happening in your case. I wil have to re produce the setup. It's been so long since I worked with this software.

eldewany, 2021-10-05:
@mzahana Thanks for your efforts, waiting for your reply

eldewany, 2021-10-08:
Fixed by refreshing an installation.