mzahana/px4_fast_planner #36: Quadrotor cannot catch up with the planed path

heleidsn, 2021-11-29:
Hi,

First a big thanks for publishing this repo. It's really useful for PX4 users. 

During my test, I find that the controller cannot catch up with the planner correctly, especially when I increase the `max_vel` in `px4_kino_replan.launch` bigger than 3. 

You can see the details in the picture below:
[image]
The yellow arrows are going faster than the real quadrotor. 

I don't know if there is something wrong with my environment, or it's limited by the geometric controller and the quadrotor's ability. Could you please give me some advice about this? Thanks！

heleidsn, 2021-11-30:
Update:

When I changed the parameters of the geometric_controller (especially increase `max_acc`, `kp_x `and `kp_y`), this gap was reduced. **So the problem is related to the controller performance.** 

However, I am still curious about if it is possible to minimize the gap between the planned pose and the real pose, such as compulsory unifying them at the beginning of each path generation (about 1Hz)?

Is there any advice? Thanks.

brunopinto900, 2022-03-29:
@heleidsn One great way to contribute is to add a MPC layer on top of the nonlinear optimization framework used in Fast Planner.