mzahana/px4_fast_planner #38: Cannot locate node of type [fast_planner_node] in package [plan_manage]；cannot launch node of type [plan_manage/traj_server]

hjl0404, 2021-12-17:
Hello, thank you very much for your sharing, but I have some troubles in the process of using it, so I would like to ask you for advice.This problem has been bothering me for a few days. I am looking forward to your reply.

After I run roslaunch px4_fast_planner px4_fast_planner. Launch, the following error appears.

ERROR: cannot launch node of type [plan_manage/fast_planner_node]: Cannot locate node of type [fast_planner_node] in package [plan_manage]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [plan_manage/traj_server]: Cannot locate node of type [traj_server] in package [plan_manage]. Make sure file exists in package path and permission is set to executable (chmod +x)

Version - PX4 Firmware v1.10.1, Ubuntu 18.04, ROS melodic

This problem has been proposed and solved by someone before. I checked his method, and he solved it after deleting pcl-1.11, but I didn't find pcl-1.11, only pcl-1.8.

Here's what I added to bashrc：

px4_dir=~/Firmware
source /opt/ros/melodic/setup.bash
source /home/hjl/catkin_ws/devel/setup.bash
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/hjl/catkin_ws/src/px4_fast_planner/models
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/hjl/catkin_ws/src/px4_fast_planner

hjl0404, 2021-12-19:
I solved this problem by referring to answer in question 12.

closed 2021-12-19 by hjl0404

mzahana, 2021-12-19:
@hjl0404 Great to hear that your issues is resolved. Can you please share a link to question 12 that you mentioned?

hjl0404, 2021-12-20:
> @hjl0404 Great to hear that your issues is resolved. Can you please share a link to question 12 that you mentioned?

Sorry, this is my first github question and didn't notice this, I have associated it with question 12, please see if this satisfies the requirements.