mzahana/px4_fast_planner #40: the drone wobbles in and around the starting point

yangy-98, 2022-02-20:
Hello,
when I run the command:
`roslaunch px4_fast_planner px4_fast_planner.launch`

the terminal shows:
```
... logging to /home/njust/.ros/log/eb340e2c-9220-11ec-b2ca-f8633f2ed425/roslaunch-nuc-17513.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
started roslaunch server http://nuc:44437/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera_pose_publisher/child_frame: camera_link
 * /camera_pose_publisher/parent_frame: world
 * /camera_pose_publisher/pose_topic: camera/pose
 * /fast_planner_node/bspline/limit_acc: 2.5
 * /fast_planner_node/bspline/limit_ratio: 1.1
 * /fast_planner_node/bspline/limit_vel: 1.0
 * /fast_planner_node/fsm/act_map: False
 * /fast_planner_node/fsm/flight_type: 1
 * /fast_planner_node/fsm/thresh_no_replan: 2.0
 * /fast_planner_node/fsm/thresh_replan: 0.5
 * /fast_planner_node/fsm/waypoint0_x: 19.0
 * /fast_planner_node/fsm/waypoint0_y: 0.0
 * /fast_planner_node/fsm/waypoint0_z: 1.0
 * /fast_planner_node/fsm/waypoint1_x: 0.0
 * /fast_planner_node/fsm/waypoint1_y: 10.0
 * /fast_planner_node/fsm/waypoint1_z: 1.0
 * /fast_planner_node/fsm/waypoint2_x: -19.0
 * /fast_planner_node/fsm/waypoint2_y: 0.0
 * /fast_planner_node/fsm/waypoint2_z: 1.0
 * /fast_planner_node/fsm/waypoint_num: 2
 * /fast_planner_node/heading_planner/half_vert_num: 3
 * /fast_planner_node/heading_planner/lambda1: 2.0
 * /fast_planner_node/heading_planner/lambda2: 1.0
 * /fast_planner_node/heading_planner/max_yaw_rate: 1.04719753333
 * /fast_planner_node/heading_planner/w: 10.0
 * /fast_planner_node/heading_planner/yaw_diff: 0.0872664611111
 * /fast_planner_node/manager/clearance_threshold: 0.2
 * /fast_planner_node/manager/control_points_distance: 0.3
 * /fast_planner_node/manager/dynamic_environment: 0
 * /fast_planner_node/manager/local_segment_length: 7.0
 * /fast_planner_node/manager/max_acc: 2.5
 * /fast_planner_node/manager/max_jerk: 4.0
 * /fast_planner_node/manager/max_vel: 1.0
 * /fast_planner_node/manager/use_geometric_path: False
 * /fast_planner_node/manager/use_kinodynamic_path: False
 * /fast_planner_node/manager/use_optimization: True
 * /fast_planner_node/manager/use_topo_path: True
 * /fast_planner_node/optimization/algorithm1: 15
 * /fast_planner_node/optimization/algorithm2: 11
 * /fast_planner_node/optimization/dist0: 0.4
 * /fast_planner_node/optimization/dist1: 0.0
 * /fast_planner_node/optimization/lambda1: 10.0
 * /fast_planner_node/optimization/lambda2: 5.0
 * /fast_planner_node/optimization/lambda3: 0.0
[... 523 lines cut ...]
[world_map_linker-2] killing on exit
[gazebo_gui-6] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

the errors:
`Warning [parser.cc:950] XML Element[vertical_fov], child of element[camera] not defined in SDF. Ignoring[vertical_fov]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.`

and
```
WARN  [mavlink] IGN REQUEST LIST: Busy
WARN  [mavlink] IGN REQUEST LIST: Busy
[ INFO] [1645343094.905536831, 15.000000000]: WP: mission received
[ERROR] [1645343094.909636081, 15.004000000]: FCU: IGN REQUEST LIST: Busy
[ERROR] [1645343094.969012656, 15.056000000]: FCU: IGN REQUEST LIST: Busy
[ WARN] [1645343096.506352936, 16.000000000]: RP: timeout, retries left 2
[ WARN] [1645343096.506594236, 16.000000000]: GF: timeout, retries left 2
WARN  [mavlink] IGN REQUEST LIST: Busy
```

I have no idea with these problems. How to solve it? I look forward to your reply. Thank you!

yangy-98, 2022-02-20:
and in gazebo, the drone wobbles in and around the starting point.

yangy-98, 2022-02-22:
@mzahana I really need help, please

calmelo, 2022-03-20:
> Hello, when I run the command: `roslaunch px4_fast_planner px4_fast_planner.launch`
>
> the terminal shows:
>
> ```
> ... logging to [...]

Have you resolved the problem? I meet the problem too.

yangy-98, 2022-03-20:
I'm sorry, I haven't solved the problem.

yangy-98, 2022-03-20:
> Have you resolved the problem? I meet the problem too.

I'm sorry, I haven't solved the problem.

Cavalletta98, 2022-11-23:
Have someone solved it?

Bugday001, 2023-01-22:
#24