mzahana/px4_fast_planner #47: 只能起飞悬停，无法进行规划

winnnnnnd, 2023-03-31:
运行roslaunch px4_fast_planner px4_fast_planner.launch后：

窗口出现以下几处报错内容，但是目前除了不能规划路径，各个功能正常。
 INFO] [1680249873.416157607, 1814.801000000]: Waiting for home pose...
**[ERROR] [1680249873.418304, 1814.801000]: Spawn service failed. Exiting.
_WARN  [mavlink] IGN REQUEST LIST: Busy
WARN  [mavlink] IGN REQUEST LIST: Busy**_
[ INFO] [1680249873.421998360, 1814.801000000]: GF: mission received
**[ERROR] [1680249873.433384826, 1814.801000000]: FCU: IGN REQUEST LIST: Busy
[ERROR] [1680249873.433797766, 1814.801000000]: FCU: IGN REQUEST LIST: Busy**
[ INFO] [1680249873.438754969, 1814.801000000]: HP: requesting home position

**vehicle_spawn_u_5068_47445212318619876-7] process has died [pid 5168, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /home/y/catkin_ws/src/px4_fast_planner/models/iris_depth_camera/iris_depth_camera.sdf -model iris_depth_camera -x 5.0 -y 0.0 -z 0.2 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_u_5068_47445212318619876 __log:=/home/y/.ros/log/a928e882-cf9a-11ed-b6ff-1c1bb5c2f1a0/vehicle_spawn_u_5068_47445212318619876-7.log].
log file: /home/y/.ros/log/a928e882-cf9a-11ed-b6ff-1c1bb5c2f1a0/vehicle_spawn_u_5068_47445212318619876-7*.log**

[ INFO] [1680249875.786677954, 1815.801000000]: RP: mission received
**[ERROR] [1680249875.789820936, 1815.809000000]: FCU: IGN REQUEST LIST: Busy**
[ INFO] [1680249876.284217261, 1816.301000000]: Got pose! Drone Ready to be armed.

[ INFO] [1680249882.225867327, 1821.313000000]: Offboard enabled
_[**fast_planner_node-9] process has died [**pid 5184, exit code -6, cmd /home/y/catkin_ws/devel/lib/plan_manage/fast_planner_node /odom_world:=/mavros/local_position/odom /sdf_map/odom:=/mavros/local_position/odom /sdf_map/cloud:=/camera/depth /sdf_map/pose:=/camera/pose /sdf_map/depth:=/camera/depth/image_raw __name:=fast_planner_node __log:=/home/y/.ros/log/a928e882-cf9a-11ed-b6ff-1c1bb5c2f1a0/fast_planner_node-9.log].
log file: /home/y/.ros/log/a928e882-cf9a-11ed-b6ff-1c1bb5c2f1a0/fast_planner_node-9*.log_
规划节点died,请问有人遇见过吗

SC-ARLab, 2023-04-13:
得看log文件

Chai8226, 2023-04-14:
可能是内存炸了，我规划到距离障碍物近的地方就容易died。我是16gx2的内存

Sujingqiao, 2023-06-09:
模型文件得提前下载下来

~/.gazebo/models/

Sujingqiao, 2023-06-09:
~/.gazebo/models/
这个目录下 模型文件下载了吗？